flashrom/flashrom #232: Add support for W25Q16JW(NIQ)

ShaunTheQuietGamer, 2021-12-04:
Data sheet here: https://www.winbond.com/resource-files/W25Q16JW%20RevG%2006302021%20Plus.pdf

It's sometimes detected automatically as W25Q16.W, and says it's only 2M-bit when according to the data sheet it is a 16M-bit chip. It seems to correctly read/write the first 2M-bits, but cannot write a 16M-bit BIOS. using a ch341a if that matters.

Th3Fanbus, 2021-12-06:
Do you have a flashrom log? You can add the `-o logfile.txt` parameter when running flashrom to obtain a verbose log. I'm pretty sure flashrom says the chip is "2048 kB", which is 2M-Byte and not 2M-bit.

About the write problems, without a log I can only guess. The only thing I can think of is that the chip you're trying to flash uses 1.8V (absolute maximum is 2.5V), but the CH341A provides 3.3V. So, you have to use a level shifter to adapt the voltages accordingly, such as this one:

[image]

Without using an adapter, the flash chip won't be happy: it will get very hot and can stop working. Even if it survives, its reliability might be compromised. I've made this mistake in the past and only noticed it after flashrom repeatedly failed to verify the flash chip's contents after writing.

ShaunTheQuietGamer, 2021-12-08:
I am already using a 1.8v adapter. No I didn't grab a log, I'll see if i can get one in a few days, its a chore to set everything up and I clearly didn't get the best ch341, it has decided to shed what I'm thinking is fiberglass (can't actually see it, just feel it,) and its not the nicest to have stabbing me all over my hands, so I'm going to have to be more careful to not handle stuff without gloves, which means I have to have a large block of time to prepare. electronic storage units are confusing (no idea what the proper term is, probably because I'm half asleep.) I didn't even know bits existed for the longest time, I finally learned about them when someone explained the bs isp's use to confuse unsuspecting victims.