flashrom/flashrom #233: Implicit conversion error in libflashrom.c

JustinBa, 2021-12-06:
Hello, was trying to build this program from source as Ubuntu is unable to locate the package using "sudo apt install flashrom" and I ran into an implicit conversion error in libflashrom.c. Exact error message is as follows.

cc -MMD -Os -Wall -Wextra -Wno-unused-parameter -Wshadow -Wmissing-prototypes -Wwrite-strings -Werror -I/usr/include/libusb-1.0  -D'CONFIG_DEFAULT_PROGRAMMER=PROGRAMMER_INVALID' -D'CONFIG_DEFAULT_PROGRAMMER_ARGS="''"' -D'CONFIG_INTERNAL=1' -D'CONFIG_INTERNAL_DMI=1' -D'CONFIG_SERPROG=1' -D'CONFIG_RAYER_SPI=1' -D'CONFIG_PONY_SPI=1' -D'CONFIG_BITBANG_SPI=1' -D'CONFIG_NIC3COM=1' -D'CONFIG_GFXNVIDIA=1' -D'CONFIG_SATASII=1' -D'CONFIG_ATAVIA=1' -D'CONFIG_IT8212=1'   -D'CONFIG_PICKIT2_SPI=1' -D'CONFIG_STLINKV3_SPI=1'   -D'CONFIG_DUMMY=1' -D'CONFIG_DRKAISER=1' -D'CONFIG_NICREALTEK=1' -D'CONFIG_NICINTEL=1' -D'CONFIG_NICINTEL_SPI=1' -D'CONFIG_NICINTEL_EEPROM=1' -D'CONFIG_OGP_SPI=1' -D'CONFIG_BUSPIRATE_SPI=1' -D'CONFIG_DEDIPROG=1' -D'CONFIG_DEVELOPERBOX_SPI=1' -D'CONFIG_SATAMV=1' -D'CONFIG_LINUX_MTD=1' -D'CONFIG_LINUX_SPI=1' -D'CONFIG_CH341A_SPI=1' -D'CONFIG_DIGILENT_SPI=1' -D'NEED_PCI=1'  -D'NEED_RAW_ACCESS=1' -D'NEED_LIBUSB1=1' -D'HAVE_UTSNAME=1' -D'HAVE_CLOCK_GETTIME=1' -D'FLASHROM_VERSION="v1.2"' -o libflashrom.o -c libflashrom.c
libflashrom.c: In function 'flashrom_supported_boards':
libflashrom.c:191:53: error: implicit conversion from 'enum test_state' to 'enum flashrom_test_state' [-Werror=enum-conversion]
  191 |                         supported_boards[i].working = binfo[i].working;
      |                                                     ^
libflashrom.c: In function 'flashrom_supported_chipsets':
libflashrom.c:229:54: error: implicit conversion from 'enum test_state' to 'enum flashrom_test_state' [-Werror=enum-conversion]
  229 |                         supported_chipsets[i].status = chipset[i].status;
      |                                                      ^
cc1: all warnings being treated as errors
make: *** [Makefile:1146: libflashrom.o] Error 1

JustinBa, 2021-12-06:
This is with GCC 11.2.0 on Ubuntu 21.10

kruzda, 2022-04-10:
This has been fixed in the master branch since. See https://forums.puri.sm/t/coreboot-4-12-based-firmware-released-for-all-librems/9750/48

randall-mcpherson, 2023-05-21:
I had a similar issue just now. I ran make and received a bunch of text, then:
```
/usr/bin/ld: cannot find -ludev: No such file or directory
collect2: errorL ld returned 1 exit status
make: *** [Makefile:1028: flashrom] Error 1
```