flashrom/flashrom #234: [100$ bounty] add support to W25Q64JVSIQ chip

nicprov, 2021-12-17:
I'm offering a 100$ bounty to whoever can add support for the Winbond 25Q64JVSIQ chip found on the [H110M-DVS](https://www.asrock.com/mb/Intel/H110M-DVS%20R3.0/index.asp). If required, escrow can be used, and I can pay using the crypto of your choice.

Essentially, I am trying to read the SOIC8 chip, and I keep getting the following error: "No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.". After checking the [list of supported chips](https://flashrom.org/Supported_hardware), I cannot find this exact model. I tried reading the chip by specifying " -c W25Q64.V", but I got the same result.

[image]

Th3Fanbus, 2021-12-18:
Hmmm, that chip is already supported, so it's likely a problem with your programmer: RDID returns all zeroes.

nicprov, 2021-12-18:
@Th3Fanbus that's what I wasn't sure about. I tried everything really, different jumper wires, different raspberry pis, and I'm using the pamona 5250 clip. 

When I look at the [list of supported chips](https://www.flashrom.org/Supported_hardware), I only see the following which have a similar name:
[image]

They look different than my chip though (W25Q64JVSIQ) because mine is supposedly 64MB from what I can find online, not 8MB.

closed 2021-12-30 by nicprov

Th3Fanbus, 2021-12-31:
It's 64M-bit, which is 8 MB (strictly speaking, MiB). Of course vendors prefer to show the larger number.