flashrom/flashrom #235: Support for Puya P25D80H

gfrancesco, 2021-12-18:
Hi, I wonder if it's possible to add support.

Datasheet:
[2006021534_-PUYA--P25D80H-SSH-IT_C559199.pdf](https://github.com/flashrom/flashrom/files/7739816/2006021534_-PUYA--P25D80H-SSH-IT_C559199.pdf)

Log:
https://pastebin.com/g4H79Wyq

[image]

I'm using a CH341A programmer modded to use 3.3V levels.

Thanks!

Th3Fanbus, 2021-12-19:
From your log, looks like there's a wiring problem. The identification command returns different data on every subsequent invocation.

gfrancesco, 2021-12-20:
What you mean is that at the end of all the lines that start with `Probing for ...` the values of `id1 0xXX, id2 0xXXXX` should always be the same for all the probes performed?

Th3Fanbus, 2021-12-20:
Yes, exactly.

gfrancesco, 2021-12-20:
Ok thanks, I think I need to get some extra hardware, I can't firmly grab the chip with the soic-8 clip I'm using, looks like a resistor close by is in the way. Hoping it's not a programmer issue since it's the first time I'm using it.

I've read the contribution guidelines, I'll try to sort out and submit a PR when I'm ready.

Droid-MAX, 2023-05-07:
or consider trying SNANDer?

gfrancesco, 2025-06-30:
Nope sorry, I used a workaround and programmed the flash via the microcontroller.