flashrom/flashrom #237: ftdi_read_data: -1, usb bulk read failed using FT232H Adafruit breakout board

0x0ptr, 2022-01-17:
I am trying to read a Winbond flash chip "W25Q80.V" (1024 kB, SPI) using a [Adafruit FT232H breakout board](https://www.adafruit.com/product/2264). The chip is detected correctly but then it fails to read:

>flashrom -p ft2232_spi:type=232H -r W25Q80.V_1024kB.bin
>flashrom v1.2-578-g3ae04a12 on Linux 5.16.0-arch1-1 (x86_64)
>
>flashrom is free software, get the source code at https://flashrom.org
>Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
>Found Winbond flash chip "W25Q80.V" (1024 kB, SPI) on ft2232_spi.
>Reading flash... ftdi_read_data: -1, usb bulk read failed
>get_buf failed: 1
>Read operation failed!
>FAILED.

I can successfully read the chip with [ftdiflash](https://github.com/adafruit/ftdiflash) for example so I know the chip is readable.

When trying to read with flashrom I connected the chip up as described in the "Generic Pinout" section at the bottom of https://flashrom.org/FT2232SPI_Programmer
Above output as you can see was from a recent build of flashrom from git. I also tried with the latest released version of flashrom (v1.2) with the same result.

farlepet, 2022-04-17:
I get the same issue with this board. Based on https://forums.adafruit.com/viewtopic.php?f=19&p=556557, reducing the read size to 16 bytes (you can probably bump it to 18) allows it to read properly. I was able to get it to work by decreasing the 16 MiB in `spi.c::spi_chip_read()` down to 16 bytes:

```c
to_read = min(ALIGN_DOWN(start + 16, 16) - start, len);
```

Though it takes several minutes to read (in my case) a 4 MiB flash. I assume this is longer than if larger reads were used, as there is a (relatively) large gap between reads.

farlepet, 2022-04-18:
Though that's a very hacky way to do it. I'd think it would make more sense (assuming this isn't a bug elsewhere causing large reads to fail) to reduce the `max_data_{read,write}` values in the `spi_master_ft2232` struct depending on the specific interface chip. A quick test shows this also works as expected.

farlepet, 2022-04-23:
Created patch for review: https://review.coreboot.org/c/flashrom/+/63798

matsievskiysv, 2022-09-27:
I have the same problem with FT2232H