flashrom/flashrom #238: "Intel Apollo Lake"

SWUS337, 2022-01-21:
Warning: unexpected second chipset match: "Intel Apollo Lake"
ignoring, please report lspci and board URL to flashrom@flashrom.org
with 'CHIPSET: your board name' in the subject line.
Found Winbond flash chip "W25Q128.JW.DTR" (16384 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Block protection could not be disabled!
Reading flash... done.
SUCCESS
FMAP REGION: RW_LEGACY
Name                           Offset     Type           Size   Comp
payload                        0x0        simple elf      53662 none
seavgabios.rom                 0xd200     optionrom       28672 none
links                          0x14240    raw               796 none
bootorder                      0x145c0    raw               465 none
etc/boot-menu-key              0x14800    raw                 8 none
etc/boot-menu-message          0x14880    raw                28 none
etc/boot-menu-wait             0x14900    raw                 8 none
(empty)                        0x14980    null          2012696 none

Th3Fanbus, 2022-01-22:
Is this log from upstream flashrom? Looks like Chromium/Chrome OS flashrom.

SWUS337, 2022-01-23:
Hello,

Short answer is (I believe) Chromium/Chrome OS from the SSH (ctrl+shift+t)
terminal in developer mode.

Should there be anything else I can do from my end to provide additional
information just let me know.

I had been playing with both the Linux and SSH (ctrl+shift+t) terminals
looking to increase functionality while exploring what might be the most
intuitive build environment for me.

I'd meant to add more to the issue however as I was also messing with
Chrome flags I crashed the system.

Honestly, part time I've been learning (teaching myself) to code and use
commands in an effort to better move forward the shared vision of Sovereign
Wealth US. It's simply not within my means to continue paying for further
development or a webmaster.

Kind regards,

Lee

Th3Fanbus, 2022-01-24:
Right, so it's most likely Chrome OS flashrom. Upstream flashrom doesn't have this problem. Just to be sure, it would be nice if you could provide a full log.