flashrom/flashrom #245: Simplify disabling incompatible programmers on Windows

obskyr, 2022-02-17:
## Background
As it stands, there are four categories of programmer that are unsupported by Flashrom on Windows. They are:

* Linux-specific programmers: `linux_mtd` and `linux_spi`
* PCI programmers: `atahpt`, `atapromise`, `atavia`, `drkaiser`, `gfxnvidia`, `internal`, `it8212`, `nic3com`, `nicintel`, `nicintel_eeprom`, `nicintel_spi`, `nicnatsemi`, `nicrealtek`, `ogp_spi`, `satamv`, and `satasii`
* Programmers that need x86 MSR or port IO: `internal`, `atahpt`, `atapromise`, `nic3com`, `nicnatsemi`, `nicrealtek`, `rayer_spi`, and `satamv`
* I²C programmers: `lspcon_i2c_spi`, `mediatek_i2c_spi`, and `realtek_mst_i2c_spi`

Perhaps some of these should be supported in the future! However, at the moment, they're incompatible with Windows.

## The problem
In the latest stable version of Flashrom, v1.2, it sufficed to specify `-Dpciutils=false`, `-Dconfig_linux_mtd=false`, and `-Dconfig_linux_spi=false`. Disabling `pciutils` automatically disabled PCI programmers. However, when building the latest commit (b9e8b89), it requires you to **manually disable each programmer** – otherwise it'll error out, either saying [“one or more enabled programmer requires PCI access”](https://github.com/flashrom/flashrom/blob/b9e8b898c03a0ee5a6d54ab9f1b2910fe75559ea/meson.build#L177-L179), or producing a mysterious compiler or linker error. Thus, **the command currently required to build at all** on Windows is the unwieldy:

```
meson \
    -Dpciutils=false \
    -Dconfig_atahpt=false \
    -Dconfig_atapromise=false \
    -Dconfig_atavia=false \
    -Dconfig_drkaiser=false \
    -Dconfig_gfxnvidia=false \
    -Dconfig_internal=false \
    -Dconfig_it8212=false \
    -Dconfig_linux_mtd=false \
    -Dconfig_linux_spi=false \
    -Dconfig_lspcon_i2c_spi=false \
    -Dconfig_mediatek_i2c_spi=false \
    -Dconfig_nic3com=false \
    -Dconfig_nicintel=false \
    -Dconfig_nicintel_eeprom=false \
    -Dconfig_nicintel_spi=false \
    -Dconfig_nicnatsemi=false \
    -Dconfig_nicrealtek=false \
    -Dconfig_ogp_spi=false \
    -Dconfig_rayer_spi=false \
    -Dconfig_realtek_mst_i2c_spi=false \
    -Dconfig_satamv=false \
    -Dconfig_satasii=false \
    --prefix="${MINGW_PREFIX}" \
    --wrap-mode=nodownload \
    --auto-features=enabled \
    --buildtype=plain \
    build

meson compile
```

I personally wrestled with this for days before I managed to find a combination of options that worked (creating #243 and #244, now closed, in the process). It's clear that it's not the most user-friendly build flow!

## The fix

I suggest doing one of the following:
* Automatically disable incompatible programmers when the system is detected to be Windows.
* Provide override options to disable whole categories at once (say, `-Dconfig_no_libpci=true`, etc., or even `-Dconfig_windows_programmers_only=true`).

Which would you rather be implemented?

## Build instructions

To build on Windows to test this, go through the following steps:

1. Install [MSYS2](https://www.msys2.org/) on Windows.
2. Open the “MSYS2 MinGW x64” shell.
3. Run `pacman -S git base-devel pactoys`.
4. Run `pacboy -S meson cc pkg-config libftdi libusb`.
5. Clone the repo and `cd` into it: `git clone https://github.com/flashrom/flashrom.git && cd flashrom`.
6. Run the following command to attempt to build:
   ```bash
   meson \
       -Dpciutils=false \
       -Dconfig_linux_mtd=false \
       -Dconfig_linux_spi=false \
       -Dconfig_atahpt=false \
       -Dconfig_atapromise=false \
       -Dconfig_atavia=false \
       -Dconfig_drkaiser=false \
       -Dconfig_gfxnvidia=false \
       -Dconfig_internal=false \
       -Dconfig_it8212=false \
       -Dconfig_nic3com=false \
       -Dconfig_nicintel=false \
       -Dconfig_nicintel_eeprom=false \
       -Dconfig_nicintel_spi=false \
       -Dconfig_nicnatsemi=false \
       -Dconfig_nicrealtek=false \
       -Dconfig_ogp_spi=false \
       -Dconfig_rayer_spi=false \
       -Dconfig_satamv=false \
       -Dconfig_satasii=false \
       --prefix="${MINGW_PREFIX}" \
       --wrap-mode=nodownload \
       --auto-features=enabled \
       --buildtype=plain \
       ./build

   cd build && meson compile
   ```

obskyr, 2022-02-22:
It has come to my attention that PCI programmers might be supported on Windows using `pciutils` – is this true? Might someone have a more definitive idea of which of these categories aren't supported on Windows? The x86 MSR and port IO ones are definitely not supported, and from what I understand the `linux_…` ones are OS-specific as well – but are the PCI ones supported on Windows, then? And how about I²C?

obskyr, 2022-02-22:
That all these categories of programmer are unsupported on Windows is based on an understanding cobbled together from (potentially outdated) documentation and code-skimming, so if someone could confirm or deny that I've got the correct list, that would be nice as well!

BATMAN83020, 2026-10-01:
Cosa cazzo dici deficiente impiccione, io sono un'esperto professionista e uso Windows x64 e non ho nessun problema con alcuni di questi programmatori, se non li sai usare o magari sei uno dei tanti utonti sbruffoni che usano quelle porcherie USB tipo imbecillino (Arduino, ecc.) non rompere i coglioni e fai quello che sai fare perché se fosse per me ti leverei pure l'ossigeno per respirare, testa di cazzo, vai a scavarti la fossa invece che dire puttanate su cose che non ti riguardano, impiccione di merda schifoso ....