flashrom/flashrom #246: Request support for RISC-V architecture platform

taotieren, 2022-02-26:
I encountered an architecture unsupported error when I tried to build flashrom on archlinux-riscv
```bash
❯ rv64build
==> 获取源代码...
  -> 找到 flashrom-v1.2.tar.bz2
  -> 找到 flashrom-v1.2.tar.bz2.asc
==> 正在为源代码文件创建完整性检查值...
[sudo] password for taotieren: 
:: Synchronizing package databases...
 core downloading...
 extra downloading...
 community downloading...
:: Starting full system upgrade...
resolving dependencies...
looking for conflicting packages...

Package (1)     Old Version  New Version  Net Change  Download Size

core/libxcrypt  4.4.28-1     4.4.28-2       0.00 MiB       0.09 MiB

Total Download Size:   0.09 MiB
Total Installed Size:  0.18 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 libxcrypt-4.4.28-2-riscv64 downloading...
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
:: Processing package changes...
upgrading libxcrypt...
==> Building in chroot for [extra] (riscv64)...
==> Synchronizing chroot copy [/var/lib/archbuild/extra-riscv64/root] -> [taotieren]...done
==> Making package: flashrom 1.2-2 (Sat Feb 26 16:06:54 2022)
==> Retrieving sources...
  -> Found flashrom-v1.2.tar.bz2
  -> Found flashrom-v1.2.tar.bz2.asc
==> Validating source files with sha256sums...
    flashrom-v1.2.tar.bz2 ... Passed
    flashrom-v1.2.tar.bz2.asc ... Skipped
==> Verifying source file signatures with gpg...
    flashrom-v1.2.tar.bz2 ... Passed
==> Making package: flashrom 1.2-2 (Sat 26 Feb 2022 04:07:07 PM CST)
==> Checking runtime dependencies...
==> Installing missing dependencies...
resolving dependencies...
looking for conflicting packages...

Package (7)          New Version  Net Change

community/confuse    3.3-3          0.12 MiB
core/hwdata          0.356-1        1.98 MiB
core/kmod            29-1           0.21 MiB
core/libusb          1.0.25-2       0.18 MiB
community/libftdi    1.5-4          0.47 MiB
extra/libusb-compat  0.1.7-1        0.03 MiB
core/pciutils        3.7.0-2        0.24 MiB

Total Installed Size:  3.24 MiB

:: Proceed with installation? [Y/n] 
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
:: Processing package changes...
installing libusb...
installing confuse...
[... 137 lines cut ...]
cc -Ilibflashrom.so.1.0.0.p -I. -I../flashrom-v1.2 -I/usr/include/libusb-1.0 -I/usr/include/libftdi1 -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c99 -Wwrite-strings -Wno-unused-parameter -Wno-address-of-packed-member -Wno-enum-conversion -Wno-missing-braces -D_DEFAULT_SOURCE -D_POSIX_C_SOURCE -D_BSD_SOURCE '-DFLASHROM_VERSION="v1.2"' -DHAVE_CLOCK_GETTIME=1 -DHAVE_STRNLEN=1 -DHAVE_UTSNAME=1 -march=rv64gc -mabi=lp64d -O2 -pipe -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -flto -fPIC -DNEED_PCI=1 -DCONFIG_ATAVIA=1 -DCONFIG_BUSPIRATE_SPI=1 -DCONFIG_CH341A_SPI=1 -DCONFIG_DEDIPROG=1 -DCONFIG_DEVELOPERBOX_SPI=1 -DCONFIG_DIGILENT_SPI=1 -DCONFIG_DRKAISER=1 -DCONFIG_DUMMY=1 -DCONFIG_FT2232_SPI=1 -DHAVE_FT232H=1 -DCONFIG_GFXNVIDIA=1 -DCONFIG_INTERNAL=1 -DCONFIG_INTERNAL_DMI=1 -DCONFIG_IT8212=1 -DCONFIG_LINUX_MTD=1 -DCONFIG_LINUX_SPI=1 -DCONFIG_NIC3COM=1 -DCONFIG_NICINTEL=1 -DCONFIG_NICINTEL_EEPROM=1 -DCONFIG_NICINTEL_SPI=1 -DCONFIG_NICREALTEK=1 -DCONFIG_OGP_SPI=1 -DCONFIG_PICKIT2_SPI=1 -DCONFIG_PONY_SPI=1 -DCONFIG_RAYER_SPI=1 -DCONFIG_SATAMV=1 -DCONFIG_SATASII=1 -DCONFIG_SERPROG=1 -DCONFIG_USBBLASTER_SPI=1 -DCONFIG_STLINKV3_SPI=1 -DCONFIG_BITBANG_SPI=1 -DNEED_RAW_ACCESS=1 -MD -MQ libflashrom.so.1.0.0.p/atavia.c.o -MF libflashrom.so.1.0.0.p/atavia.c.o.d -o libflashrom.so.1.0.0.p/atavia.c.o -c ../flashrom-v1.2/atavia.c
In file included from ../flashrom-v1.2/atavia.c:23:
../flashrom-v1.2/hwaccess.h:300:2: error: #error Unknown architecture, please check if it supports PCI port IO.
  300 | #error Unknown architecture, please check if it supports PCI port IO.
      |  ^~~~~
[3/140] cc -Ilibflashrom.so.1.0.0.p -I. -I../flashrom-v1.2 -I/usr/include/libusb-1.0 -I/usr/include/libftdi1 -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c99 -Wwrite-strings -Wno-unused-parameter -Wno-address-of-packed-member -Wno-enum-conversion -Wno-missing-braces -D_DEFAULT_SOURCE -D_POSIX_C_SOURCE -D_BSD_SOURCE '-DFLASHROM_VERSION="v1.2"' -DHAVE_CLOCK_GETTIME=1 -DHAVE_STRNLEN=1 -DHAVE_UTSNAME=1 -march=rv64gc -mabi=lp64d -O2 -pipe -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -flto -fPIC -DNEED_PCI=1 -DCONFIG_ATAVIA=1 -DCONFIG_BUSPIRATE_SPI=1 -DCONFIG_CH341A_SPI=1 -DCONFIG_DEDIPROG=1 -DCONFIG_DEVELOPERBOX_SPI=1 -DCONFIG_DIGILENT_SPI=1 -DCONFIG_DRKAISER=1 -DCONFIG_DUMMY=1 -DCONFIG_FT2232_SPI=1 -DHAVE_FT232H=1 -DCONFIG_GFXNVIDIA=1 -DCONFIG_INTERNAL=1 -DCONFIG_INTERNAL_DMI=1 -DCONFIG_IT8212=1 -DCONFIG_LINUX_MTD=1 -DCONFIG_LINUX_SPI=1 -DCONFIG_NIC3COM=1 -DCONFIG_NICINTEL=1 -DCONFIG_NICINTEL_EEPROM=1 -DCONFIG_NICINTEL_SPI=1 -DCONFIG_NICREALTEK=1 -DCONFIG_OGP_SPI=1 -DCONFIG_PICKIT2_SPI=1 -DCONFIG_PONY_SPI=1 -DCONFIG_RAYER_SPI=1 -DCONFIG_SATAMV=1 -DCONFIG_SATASII=1 -DCONFIG_SERPROG=1 -DCONFIG_USBBLASTER_SPI=1 -DCONFIG_STLINKV3_SPI=1 -DCONFIG_BITBANG_SPI=1 -DNEED_RAW_ACCESS=1 -MD -MQ libflashrom.so.1.0.0.p/pcidev.c.o -MF libflashrom.so.1.0.0.p/pcidev.c.o.d -o libflashrom.so.1.0.0.p/pcidev.c.o -c ../flashrom-v1.2/pcidev.c
FAILED: libflashrom.so.1.0.0.p/pcidev.c.o 
cc -Ilibflashrom.so.1.0.0.p -I. -I../flashrom-v1.2 -I/usr/include/libusb-1.0 -I/usr/include/libftdi1 -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c99 -Wwrite-strings -Wno-unused-parameter -Wno-address-of-packed-member -Wno-enum-conversion -Wno-missing-braces -D_DEFAULT_SOURCE -D_POSIX_C_SOURCE -D_BSD_SOURCE '-DFLASHROM_VERSION="v1.2"' -DHAVE_CLOCK_GETTIME=1 -DHAVE_STRNLEN=1 -DHAVE_UTSNAME=1 -march=rv64gc -mabi=lp64d -O2 -pipe -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -flto -fPIC -DNEED_PCI=1 -DCONFIG_ATAVIA=1 -DCONFIG_BUSPIRATE_SPI=1 -DCONFIG_CH341A_SPI=1 -DCONFIG_DEDIPROG=1 -DCONFIG_DEVELOPERBOX_SPI=1 -DCONFIG_DIGILENT_SPI=1 -DCONFIG_DRKAISER=1 -DCONFIG_DUMMY=1 -DCONFIG_FT2232_SPI=1 -DHAVE_FT232H=1 -DCONFIG_GFXNVIDIA=1 -DCONFIG_INTERNAL=1 -DCONFIG_INTERNAL_DMI=1 -DCONFIG_IT8212=1 -DCONFIG_LINUX_MTD=1 -DCONFIG_LINUX_SPI=1 -DCONFIG_NIC3COM=1 -DCONFIG_NICINTEL=1 -DCONFIG_NICINTEL_EEPROM=1 -DCONFIG_NICINTEL_SPI=1 -DCONFIG_NICREALTEK=1 -DCONFIG_OGP_SPI=1 -DCONFIG_PICKIT2_SPI=1 -DCONFIG_PONY_SPI=1 -DCONFIG_RAYER_SPI=1 -DCONFIG_SATAMV=1 -DCONFIG_SATASII=1 -DCONFIG_SERPROG=1 -DCONFIG_USBBLASTER_SPI=1 -DCONFIG_STLINKV3_SPI=1 -DCONFIG_BITBANG_SPI=1 -DNEED_RAW_ACCESS=1 -MD -MQ libflashrom.so.1.0.0.p/pcidev.c.o -MF libflashrom.so.1.0.0.p/pcidev.c.o.d -o libflashrom.so.1.0.0.p/pcidev.c.o -c ../flashrom-v1.2/pcidev.c
In file included from ../flashrom-v1.2/pcidev.c:22:
../flashrom-v1.2/hwaccess.h:300:2: error: #error Unknown architecture, please check if it supports PCI port IO.
  300 | #error Unknown architecture, please check if it supports PCI port IO.
      |  ^~~~~
[4/140] cc -Ilibflashrom.so.1.0.0.p -I. -I../flashrom-v1.2 -I/usr/include/libusb-1.0 -I/usr/include/libftdi1 -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c99 -Wwrite-strings -Wno-unused-parameter -Wno-address-of-packed-member -Wno-enum-conversion -Wno-missing-braces -D_DEFAULT_SOURCE -D_POSIX_C_SOURCE -D_BSD_SOURCE '-DFLASHROM_VERSION="v1.2"' -DHAVE_CLOCK_GETTIME=1 -DHAVE_STRNLEN=1 -DHAVE_UTSNAME=1 -march=rv64gc -mabi=lp64d -O2 -pipe -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -flto -fPIC -DNEED_PCI=1 -DCONFIG_ATAVIA=1 -DCONFIG_BUSPIRATE_SPI=1 -DCONFIG_CH341A_SPI=1 -DCONFIG_DEDIPROG=1 -DCONFIG_DEVELOPERBOX_SPI=1 -DCONFIG_DIGILENT_SPI=1 -DCONFIG_DRKAISER=1 -DCONFIG_DUMMY=1 -DCONFIG_FT2232_SPI=1 -DHAVE_FT232H=1 -DCONFIG_GFXNVIDIA=1 -DCONFIG_INTERNAL=1 -DCONFIG_INTERNAL_DMI=1 -DCONFIG_IT8212=1 -DCONFIG_LINUX_MTD=1 -DCONFIG_LINUX_SPI=1 -DCONFIG_NIC3COM=1 -DCONFIG_NICINTEL=1 -DCONFIG_NICINTEL_EEPROM=1 -DCONFIG_NICINTEL_SPI=1 -DCONFIG_NICREALTEK=1 -DCONFIG_OGP_SPI=1 -DCONFIG_PICKIT2_SPI=1 -DCONFIG_PONY_SPI=1 -DCONFIG_RAYER_SPI=1 -DCONFIG_SATAMV=1 -DCONFIG_SATASII=1 -DCONFIG_SERPROG=1 -DCONFIG_USBBLASTER_SPI=1 -DCONFIG_STLINKV3_SPI=1 -DCONFIG_BITBANG_SPI=1 -DNEED_RAW_ACCESS=1 -MD -MQ libflashrom.so.1.0.0.p/usbdev.c.o -MF li [... cut ...] RAYER_SPI=1 -DCONFIG_SATAMV=1 -DCONFIG_SATASII=1 -DCONFIG_SERPROG=1 -DCONFIG_USBBLASTER_SPI=1 -DCONFIG_STLINKV3_SPI=1 -DCONFIG_BITBANG_SPI=1 -DNEED_RAW_ACCESS=1 -MD -MQ libflashrom.so.1.0.0.p/dediprog.c.o -MF libflashrom.so.1.0.0.p/dediprog.c.o.d -o libflashrom.so.1.0.0.p/dediprog.c.o -c ../flashrom-v1.2/dediprog.c
ninja: build stopped: subcommand failed.
==> ERROR: A failure occurred in build().
    Aborting...
==> ERROR: Build failed, check /var/lib/archbuild/extra-riscv64/taotieren/build

```

felixsinger, 2022-07-02:
Could you try compiling flashrom from master please?

taotieren, 2022-07-02:
> Could you try compiling flashrom from master please?

Yes, I encountered some compilation errors and dependency problems when I tried to compile.

felixonmars, 2022-09-19:
As of current master https://github.com/flashrom/flashrom/commit/1f65e9029b8c9d0fbe6c60d8fee249c57fd3f3fa meson build on RISC-V succeeds here. Invoking `make` would still result in similar failures in OP.

felixsinger, 2022-09-19:
Yes, RISC-V is supported by the Meson build system.

We are about to move to Meson entirely. Most likely the Make build system will be deprecated with the next release. There is still some work left on our Meson integration and we also don't have a date for the release yet. However, Meson might just work for you.

closed 2022-09-23 by felixsinger