flashrom/flashrom #248: cli_classic.c:718:13: error: format '%ld' expects argument of type 'long int', but argument 2 has type 'size_t' {aka 'unsigned int'}

kevin-olbrich, 2022-03-09:
Compiling on Raspberry Pi using Debian Bullseye fails with this error:

```
root@raspberrypi:~/flashrom# make
Replacing all version templates with v1.2-642-g4bee256.
C compiler found: cc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110
Target arch: arm
Target OS: Linux
Target endian: little
Dependency libpci found: yes
  Checking for old "pci_get_dev()": no
  CFLAGS: -I/usr/include/arm-linux-gnueabihf
  LDFLAGS: -lpci -lz -lresolv -ludev
Dependency libusb1 found: no
Dependency libjaylink found: no
Dependency NI-845x found: no
Dependency libftdi1 found: no
Checking for header "mtd/mtd-user.h": yes
Checking for header "linux/spi/spidev.h": yes
Checking for header "linux/i2c-dev.h": yes
Checking for header "linux/i2c.h": yes
Checking for header "sys/utsname.h": yes
Checking for function "clock_gettime": yes
Checking for external "librt": yes
cc -MMD -Os -Wall -Wextra -Wno-unused-parameter -Wshadow -Wmissing-prototypes -Wwrite-strings -Werror -I/usr/include/arm-linux-gnueabihf   -D'IS_WINDOWS=0' -D'__FLASHROM_LITTLE_ENDIAN__=1' -D'USE_IOPL=1' -D'USE_DEV_IO=0' -D'USE_IOPERM=0' -D'CONFIG_DEFAULT_PROGRAMMER_NAME=NULL' -D'CONFIG_DEFAULT_PROGRAMMER_ARGS=""' -D'CONFIG_INTERNAL=1' -D'CONFIG_INTERNAL_DMI=1' -D'CONFIG_SERPROG=1' -D'CONFIG_PONY_SPI=1' -D'CONFIG_BITBANG_SPI=1' -D'CONFIG_GFXNVIDIA=1' -D'CONFIG_SATASII=1' -D'CONFIG_ATAVIA=1' -D'CONFIG_IT8212=1' -D'CONFIG_DUMMY=1' -D'CONFIG_DRKAISER=1' -D'CONFIG_NICINTEL=1' -D'CONFIG_NICINTEL_SPI=1' -D'CONFIG_NICINTEL_EEPROM=1' -D'CONFIG_OGP_SPI=1' -D'CONFIG_BUSPIRATE_SPI=1' -D'CONFIG_LINUX_MTD=1' -D'CONFIG_LINUX_SPI=1' -D'NEED_PCI=1' -D'HAVE_UTSNAME=1' -D'HAVE_CLOCK_GETTIME=1' -D'FLASHROM_VERSION="v1.2-642-g4bee256"' -o cli_classic.o -c cli_classic.c
cli_classic.c: In function 'main':
cli_classic.c:718:13: error: format '%ld' expects argument of type 'long int', but argument 2 has type 'size_t' {aka 'unsigned int'} [-Werror=format=]
  718 |   printf("%ld\n", flashrom_flash_getsize(fill_flash));
      |           ~~^     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             |     |
      |             |     size_t {aka unsigned int}
      |             long int
      |           %d
cc1: all warnings being treated as errors
make: *** [Makefile:968: cli_classic.o] Fehler 1
```

Th3Fanbus, 2022-03-09:
Hi, https://review.coreboot.org/62707 should fix this problem.

kevin-olbrich, 2022-03-10:
I can confirm this is fixing the problem.

Craig-Powell, 2024-12-13:
I'm running into the same message, different file, during compile on a Raspberry Pi 2 running 32-bit bookworm.  I had no issue with compiling on a Raspberry Pi 3 running 64-bit bookworm though.  The compile is of the recently released flashrom v1.5, both RPis were fresh OS images with only the necessary programs installed per the flashrom compiling directions.  Is there something I'm missing or is flashrom no longer 32-bit compatible?

```
craig@flashpi01:~ $ meson compile -C flashrombld
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/craig/flashrombld
ninja: Entering directory `/home/craig/flashrombld'
[11/96] Compiling C object tests/flashrom_unit_tests.p/chip.c.o
FAILED: tests/flashrom_unit_tests.p/chip.c.o 

...

../flashrom-v1.5.0/tests/chip.c: In function ‘progress_callback’:
../flashrom-v1.5.0/tests/chip.c:94:74: error: format ‘%ld’ expects argument of type ‘long int’, but argument 3 has type ‘size_t’ {aka ‘unsigned int’} [-Werror=format=]
   94 |                 printf("Progress almost complete for stage %d, current %ld, total %ld\n",
      |                                                                        ~~^
      |                                                                          |
      |                                                                          long int
      |                                                                        %d
   95 |                         flash->progress_state->stage, flash->progress_state->current, flash->progress_state->total);
      |                                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                            |
      |                                                                            size_t {aka unsigned int}
../flashrom-v1.5.0/tests/chip.c:94:85: error: format ‘%ld’ expects argument of type ‘long int’, but argument 4 has type ‘size_t’ {aka ‘unsigned int’} [-Werror=format=]
   94 |                 printf("Progress almost complete for stage %d, current %ld, total %ld\n",
      |                                                                                   ~~^
      |                                                                                     |
      |                                                                                     long int
      |                                                                                   %d
   95 |                         flash->progress_state->stage, flash->progress_state->current, flash->progress_state->total);
      |                                                                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                                            |
      |                                                                                                            size_t {aka unsigned int}
cc1: all warnings being treated as errors
[16/96] Generating doc/man-pages with a custom command
ninja: build stopped: subcommand failed.
craig@flashpi01:~ $ 
```
And thank you for all the developers that work on this project!   Flashrom recently saved me from a problematic BIOS flash and got the motherboard running again.

Craig-Powell, 2024-12-16:
Looks like I got past this by making the same change that was done in cli_classic.c, by changing the two %ld variables on line 94 of chip.c to %zu.  Or at least I got it to compile and then was able to run the meson test and meson install commands.  I don't have much to test with on the RPi2, and I'm having problems reading/writing to the FM25Q04 chip I'm testing with but I suspect that is with my hardware and not flashrom.